fair-acc/gnuradio4 #51: sizeof(SingleThreadedStrategy) should be minimized

mattkretz, 2023-03-02:
`SingleThreadedStrategy` puts all its data members on their own cache line. Why? It's 320 Bytes instead of 40 Bytes. Most member functions need to access `_size` and `_nextValue` / `_cachedValue`. Thus it would be helpful to have them all in the same cache line.
Proposal align `SingleThreadedStrategy` to a cache line, but not its individual members.

mattkretz, 2023-03-02:
The same question applies to `circular_buffer::buffer_impl`, `circular_buffer::buffer_writer`, and `circular_buffer::buffer_reader`. Why are those members on their own cache lines?

mattkretz, 2023-03-02:
Actually, `MultiThreadedStrategy` also shouldn't need each data member on its own cache line. What's the reasoning here?

RalphSteinhagen, 2023-03-02:
Most of the `alignas(kCacheLine)`s are to prevent false-sharing notable the atomic sequence counter.

But indeed, we should revisit some of the overzealous ones.

mattkretz, 2023-03-02:
Sure, I'm implicitly asking about what scenarios of false sharing you had in mind. Because I can't see any. This is wasting bandwidth, TLB and cache space for no apparent gain.
Note that a pointer (or reference) to `Sequence` points to an atomic that is still on its own cache line. Since the pointer isn't modified, there would be no issue even if it were accessed from multiple threads at the same time.
The rule of thumb here: if it's not atomic or protected by a mutex, then there's no issue to fix wrt. false sharing.

RalphSteinhagen, 2023-03-03:
The false-sharing scenario is primarily related to the 'Sequence' and list containing those atomic counters. In a multiple-writer-multiple-reader scenario, these should not be in the same cache line. The other constraint why the 'alignas' is used was to ensure that the total aggregate size is multiples of the cache line so that the risk of back-to-back readers/writers being split/invalidating the cache is kept to a minimum.

mattkretz, 2023-03-03:
So, what do you think about f76eea9f768d5e3fc690be99192c6bc848891e1d. Follow ups 34a340dcbdb4be3c57160d870574b98e88184354 and ef60468dca5453d7bbd8ce5c29779b259bbb7553

mattkretz, 2023-03-14:
Resolved by #55

closed 2023-03-14 by mattkretz