fair-acc/gnuradio4 #69: GR4: implement selector block/node (switch matrix like run-time plumbing between input->output ports)

ivan-cukic, 2023-05-18:


RalphSteinhagen, 2023-07-15:
We (@wirew0rm, @drslebedev, @RalphSteinhagen) came up with one viable implementation option/solution for this problem that should be easy enough to implement and provide an intuitive and simple API to the users:

- The 'selector' block -- or any other block that supports a variable number of input/output ports -- would support and implement static ports (`TI1`, `TI2`, ..., `TO1`, `TO2`, etc.) as before. These ports can be arbitrarily strictly typed (<-> being fields of the node/block struct).
- In addition to static ports, the idea is to have a pair of vectors for the input and output ports. The `Port<Tx>` inside these vectors would have the same name (modulo counting index), constraints, and, notably, type (`T_IN` for input ports and `T_OUT` for output ports).

This way, the block would support the following `process_bulk(...)` signature:
```cpp
work_return_status_t
process_bulk(std::span<TI1> in1, std::span<TI2> in2, ..., std::vector<std::span<T_IN>> dyn_in, 
             std::span<TO1> out1, std::span<TO2> out2, ..., std::vector<std::span<T_OUT>> dyn_out) {...}`
```
The same signature as above but with `std::span<>` replaced with our new `ConsumableSpan<T_IN>` and `PublishableSpan<T_OUT>` concepts.

These `ConsumableSpan` and `PublishableSpan` concepts function like `std::span<T>` but with an initial maximum size. The user can `consume(size_t)` only a partial amount of the available input and commit `produce(size_t)` only a partial amount of the initially reserved samples before the `process_bulk` function is called. This introduces the flexibility for handling varying input-to-output sample-rate ratios on the fly.

Please share your thoughts, suggestions, or potential pitfalls with this approach.

RalphSteinhagen, 2023-10-05:
finished by PR: https://github.com/fair-acc/graph-prototype/pull/178

closed 2023-10-05 by RalphSteinhagen