fair-acc/gnuradio4 #125: graph-prototype: Fix issues with uT and dynamic libraries, plugins

ivan-cukic, 2023-07-14:


wirew0rm, 2023-07-17:
Some additional info from the tests which are currently not using uT for one of the following reasons:
- Clang fails to detect the tests in conjunction with our plugin systems (no testsuites are discovered)
- Additional wasm specific limitations observed in opencmw-cpp
  - WASM tests are run by nodejs which does not implement the fetch API implemented in browsers
    - current solution: run tests using the fetch API in selenium (a test framework using a remote controlled chromium browser)
    - node 18 (or 17 with experimental flags) should implement the fetch API and there are library implementations for earlier versions.  Node 18 is the version used on the CI runners.
    - to be verified if the fetch API is the only thing that was missing in nodejs compared to the browser.
  - For emscripten there is no easy way to run the server components so we have to compile and run that part natively for the integration tests
    - Difficult in cmake to support multiple targets in the same build
    - no standard way in UT to run an external binary needed for the test.

ivan-cukic, 2023-07-19:
@wirew0rm The initial aim of this task was to fix only the issue of using uT for tests which load the plugins.

The description of what else is broken or incomplete in our tests you gave should be split and moved into different tasks as they are only loosely related.

My proposal would be to create an EPIC for all things related to tests, and create these as separate tasks under it, to avoid having one big task that will contain several PRs in the end. IMO, it is better to have one PR cover several tasks, than having one task with several PRs as that doesn't communicate the progress.

frankosterfeld, 2023-11-14:
I can convert app_grc.cpp into a ut test and that seems to work just fine, with both gcc 13 and clang 17.

I see a crash with qa_plugins_test.cpp though: The static TestContext from context() is used to register builtin_multiply via `GP_REGISTER_NODE(&context().registry, builtin_multiply, double, float);`. This static TestContext instance is then destroyed before the actual test (`"GoodPlugins"_test`) is run, causing the test to crash.
I can reproduce this issue with clang 16 and 17 (gcc passes):
```
#include <boost/ut.hpp>

using namespace boost::ut;

struct A {
    int x = 123;
    ~A() { x = 0; }
};

struct B {
    int x = 0;
};

A& a() {
    static A inst;
    return inst;
}

B staticB{ .x = a().x };

const boost::ut::suite Tests = [] {

"Test"_test = [] {
    // this fails with clang 16/17: The static inst in a() is destroyed before this test is run, x is then 0
    // with gcc, inst is not destroyed yet and the test passes
    expect(eq(a().x, 123));
};

};

int main() {}
```

frankosterfeld, 2023-11-14:
I started enabling clang/porting to ut here: frank/wip-plugin-tests-ut-clang

frankosterfeld, 2023-11-14:
Unless I miss something, the PR is ready and shows that ut and plugins work with clang (17).
I haven't touched the emscripten-related issues (where I agree with Ivan that they should be a separate issue).

RalphSteinhagen, 2023-11-15:
It seems that the original issue was solved by the new UT and/or Clang compiler version. I would leave it at this for the time being. We could re-open this if this reoccurs.

closed 2023-11-15 by RalphSteinhagen