fair-acc/gnuradio4 #161: List of GR 3.10 blocks sorted by categories and priority

drslebedev, 2023-09-13:
With this issue we want to create a list of  blocks from GR 3.10 which have to be implemented in graph-prototype.
The list is sorted by categories and priority for GSI/FAIR.

Harishankar14, 2025-03-15:
Hey, i just wanted to know, some of the high priority blocks,which needed to be migrated ?

RalphSteinhagen, 2025-03-15:
@Harishankar14, great and thanks for your interest and willingness to contribute. 👍 

The choice is yours and depends on your preference and expertise. Let's start with something simple (e.g. the boolean or byte operators) to get acquainted with the new block API and then move on to something more complex. 

If you are into HAM then the AM or FM blocks might also be a good start. We deliberately left them unimplemented so that others -- like perhaps you -- could grow into the block lib concepts.

If you have some more C++ experience and thorough understanding of DFT and FFT algorithms, there are also some other `std::simd`-related project aspects that would greatly help.

As always, feel free to ask questions on the GR channels (https://matrix.to/#/#gnuradio:gnuradio.org and https://matrix.to/#/#development:gnuradio.org). 
For GR4 specifically: https://matrix.to/#/#architecture:gnuradio.org
That's often a good and perhaps better place to start. :smile: 

<small><small>*to note: 'porting' and 'migrating' implies functionality, not just copying code, and may require some implementing from scratch. We are trying to keep the core LGPLv3, so beware that your ideas/resources are compatible (most are, except GPLv3) .*</small></small>

Harishankar14, 2025-03-15:
yeah, sure, I will look into it and provide the help that the community wants ! Btw, I am there in the GNU Channels, I go by the name **OnlyDPCanSaveMe**

jameshu15869, 2025-03-19:
Would these be OOT modules based on the tutorial on the GNURadio website for [GNU Radio 3.x](https://wiki.gnuradio.org/index.php?title=Creating_C%2B%2B_OOT_with_gr-modtool)? 

I took a look at #395 and the structure looks to be different from the OOT modules on tutorial.

RalphSteinhagen, 2025-03-20:
@jameshu15869 great and thanks for your interest in contributing.

For the basic ones (i.e. those that are functionally already in GR3 and have no specific HW or external library dependency and are LGPLv3 compatible) these would be in the GR4 block-lib, i.e. go into the sources `./block/<module>` directory. There are also some OOT usage examples for blocks that cannot be shared with the core (due to licenses, IP, ...). 

However, creating blocks in GR4 is different/easier than in GR3 and **does not require a 'modtool'** or similar. 
The code is the single source of truth and some build file entries. 

The main things you need to do are:
 1. write a `.hpp' (here is also where you can add documentation, infos for the registry, and other optional infos), 
 2. add this `.hpp' file to the `./block/<module>`/CMakeLists.txt` file (so that it gets exported/installed),
 3. and include the `.hpp` in your project, e.q. your new unit test which goe into  `./block/<module>`/test/`

For now, in addition to the tutorials, the examples in the 'basic' or 'testing' modules are a good start for reading and showing the breadth of what is possible and getting an idea of the flavour of how blocks are written, tested and used.

I will try to update/write the new block-lib dev documentation.

*N.B. documentation, tutorials, people sharing their experience and helping others to get started is something where we need help ... if you are interested. Check out the GR channels mentioned above.*

codblack789, 2025-03-27:
@RalphSteinhagen  how did u build the gnuradio4 locally.. and set it up, can u explain?

RalphSteinhagen, 2025-03-28:
> [@RalphSteinhagen](https://github.com/RalphSteinhagen) how did u build the gnuradio4 locally.. and set it up, can u explain?

You already asked the question [here](https://github.com/fair-acc/gnuradio4/pull/536#issuecomment-2760368417) and got an answer. Please do not cross-post and use the GNU Radio channels for generic IDE setup/non-GR-related questions.

jameshu15869, 2025-04-01:
Seeing #536, will this project be used for GSoC? I was curious about contributing a couple blocks in my spare time but don't want to step on someone's feet

RalphSteinhagen, 2025-04-02:
@jameshu15869 you are most welcome to join and contribute. Everybody is welcome. 

We setup this as a tracker to coordinate that things are not getting implemented twice... nothing else.

Let us (@mormj, @jsallay, et all.) know what your interest, level-of-expertise, and block category[y, ies] is/are (😁) that you'd like to tackle. I'd suggest to start with an easier one to get familiar with the basic API build-system and then move to a more complex example.

mohamed-walid-fathy, 2025-04-10:
hello, I found out about your project through GSoC, Can I work on GNU Radio even if I did not send a formal proposal? I am looking for a beginner-level bug/feature as this is my first time contributing to open source.

RalphSteinhagen, 2026-11-13:
This list is largely finished, fill close this in favour of a updated list that includes only the remaining core blocks that need to be implemented prior to an official GNU Radio 4.0 release.

closed 2026-11-13 by RalphSteinhagen