fair-acc/gnuradio4 #276: GR: Allow graph topology updates while running

ivan-cukic, 2024-02-15:
We should introduce graph update requests into the messaging system - messages to request a new block to be added, a block to be replaced with another or removed.

It should also have messages for establishing port connections or disconnecting them.

RalphSteinhagen, 2024-10-03:
Please consider that these messages should be communicated also via the service (GR<->Majordomo) and external client. 

*N.B. This does not include the corresponding UI components, handled in issue https://github.com/fair-acc/opendigitizer/issues/199 *

ivan-cukic, 2024-10-28:
Related: https://github.com/fair-acc/gnuradio4/pull/448

RalphSteinhagen, 2024-11-08:
finished by https://github.com/fair-acc/opendigitizer/pull/233

closed 2024-11-08 by RalphSteinhagen