fair-acc/gnuradio4 #385: Complete instructions on how to install and start.

newmansnj, 2024-07-30:
Hello,
I have now tried several ways to start GNuradio4. I got so far that I was able to build it via https://github.com/fair-acc/gnuradio4/blob/main/DEVELOPMENT.md (including all tests that run but do not install). 
I have tried the gnu radio wiki, old lectures, other repositories and reverse engineering the code to find out how to install the software but so far without success because unfortunately there are no or only outdated instructions on how to install and start the whole thing.

RalphSteinhagen, 2024-07-31:
@newmansnj, thank you for your interest in the GNU Radio 4.0 beta version.

Please note that GNU Radio 4.0 is a header-only library and does not require traditional installation.
You have two main options for using it:

 1. **Direct Usage**: You can fork the repository and work directly within the library. This approach is useful if you plan to contribute to the core or block library and submit pull requests.

 2. **Including in Your Project**: If you prefer to use the library 'as-is' without modifying its code, you can include it in your projects via CMake using the `FetchContent_Declare(...)` statement.

If you need further assistance or specific examples of how to include the library in your project, feel free to ask.

Cheers,
Ralph

*N.B. there is a [EU GNU Radio Days](https://events.gnuradio.org/event/23/) workshop at FAIR end of August, where we provide some in-person and online (via YouTube) tutorials on how-to use GR4 and differences to GR3.*

gitXsingh, 2025-04-01:
Hi @RalphSteinhagen ,

I found multiple `README.md` files but couldn't locate a dedicated installation guide.  
Should I add the installation steps to `docs/README.md` or create a new `INSTALL.md`?  

Let me know the best location, and I’ll proceed accordingly.  

Thanks,  
@gitXsingh

gitXsingh, 2025-04-01:
@RalphSteinhagen  
This PR addresses Issue #385 . 
Could you please review when you have time?  
I'm happy to make any adjustments needed.
PR link : https://github.com/gitXsingh/gnuradio4/pull/new/add-installation-guide

RalphSteinhagen, 2025-04-02:
@gitXsingh thanks for your interest and follow-up. 

Large parts of this seem to be standard CMake instructions and is not directly related to GR4. But I understand that for people getting into GR4, CMake (and/or other C/C++ build systems) can be a hurdle. I think that a basic `INSTALL.md` might be the best option. Please note: highlight that GR4 as header-only usually does not need to be installed but can be pulled-in into any other project via fetch content. There is usually not an install required unless you need runtime polymorphism (i.e. a pre-build block library). Maybe reference to another more detailed OOT repo might be advised.

I am not an expert on this, maybe @mormj or @cafeclimber could advise and chime-in on this.

gitXsingh, 2025-04-02:
thanks for the feedback! I have updated the INSTALL.md file to provide a basic yet clear installation guide, considering GR4's header-only nature. The update ensures that new users understand how to integrate GR4 without unnecessary installation steps. If there's anything specific that should be refined or referenced differently, let me know! I appreciate the insights from @mormj  and @cafeclimber as well.

gitXsingh, 2025-05-14:
Hi all,
I have updated the INSTALL.md file to address all feedback. The guide is now more comprehensive, beginner-friendly, and covers all aspects of installation, usage, and contribution for GNU Radio 4.
You can review the changes in PR #543.
This update should fully resolve issue #385.
Please let me know if there are any further suggestions or improvements needed. Thank you for your guidance and support!

cafeclimber, 2025-05-15:
I think if you want to develop a good getting up and running guide, working through @daniestevez 's [gr4-packet-modem](https://github.com/daniestevez/gr4-packet-modem/tree/main) repo would be a solid start. He does a good job explaining how he was able to get up and running and even [gave a talk on his work](https://www.youtube.com/watch?v=1EPuhaIjxCk). I've not myself actually built and run anything with GR4 (though I am very keen to do so!!) so can't provide a ton of guidance but hope this is helpful