fair-acc/gnuradio4 #407: Improve handling of tags with the processBulk Consumable/PublishableRange API

wirew0rm, 2024-09-11:
When using the `processBulk` API, special care has to be taken when publishing samples and the corresponding tags. At the moment this works with the automatic tag propagation which has some shortcomings:
- can only correctly forward tags on the first sample
- difficult to disable/override
To fix these, the ` Consumable/PublishablePortRange` API should be extended to allow for an intuitive way to control the propagation of tags. There was some initial offline discussion which is persisted in this issue involving @RalphSteinhagen, @drslebedev and @wirew0rm, but others are of course invited to chime in if they have a specific usecase or improvement in mind. The Proposed changes are roughly structured into four individual steps that can be tackled more or less independently.

- [ ] I: convenience methods for consuming tags in `ConsumableInputPortRange` (Concept: `ConsumablePortSpan`)
  -  at the moment one has to directly interact with the `PublishableInputRange<Tag>` which is a member of the PortRange
  - [ ] implement `consumableInputPortRange::consume(index=size::MAX)`, which will consume all tags up to the given sample index (= all tags belonging to samples in the range)
- [ ] II: Automatic Tag Propagation Policies
  -  optional block attribute (like e.g. `Resampling` or `BlockingIO`)
  - different policies:
    - back-propagate: treat the tag as if it was belonging to the first sample of the next process Bulk invocation
    - forward-propagate: treat the tag as if it were on the first sample of the current processBulk invocation
    - [ignore]: drop samples that cannot be treated on the first sample (not sure if there is a usecase for that, but only way to ensure that tags are never shifted to an earlier or later sample)
    - [keep at original position]: publish the sample at the same sample position (possible considering rate changes). Settings might only be updated later, so the processing of the chunk will still use the outdated values.
- [ ] III: convenience methods for publishing Tags in PublishablePortRange
  - [ ] implement  `PublishableOutputPortRange`
  - [ ] PublishableOutputPortRange::publishTag(index = 0)` 
  - relative and/or data-span local index
- [ ] IV: forwarding policy
  - at the moment we have all-to-all and one-to-one strategies defined, but only all-to-all is implemented
  - [We discussed providing a customization point where users could plug in predefined or their own strageies -> abandoned]
  - Instead this should all be done in the custom block processing function.
  - [ ] Implement a way to suppress the automatic tag forwarding
  - [ ] Modify the existing selector block to demonstrate how to explicitly forward the correct tags to the correct output samples.

RalphSteinhagen, 2024-09-12:
@daniestevez one of the initial design choices was to keep `processBulk(...)` simple and to only export `std::span<const T>` and `std::span<T>` interfaces. This required the tag-forwarding to be external to the processing function. This also because of samples and tags being propagated in different buffers for performance reasons.

This has been softened by adding an opt-in API that also allows to dynamically control of how many samples are consumed/produced via the `ConsumableSpan` and `ProducableSpan`.

The proposal above would extend these to `Consumable/PublishablePortRange`, which further extends these to the `Tag` handling and adds the synchronisation primitives between the sample/data and the `Tag` buffer. This way, there'd be two default propagation policies (i.e. forward and backwards) and an opt-out. The latter for users that want full control how Tags are being propagated between different ports.

drslebedev, 2024-10-28:
See follow-up issue https://github.com/fair-acc/gnuradio4/issues/407

closed 2024-10-28 by drslebedev