fair-acc/gnuradio4 #488: Subgraph's YAML-Based definition and plugin-like extension

drslebedev, 2024-12-17:
Enable the creation of new block functionalities by combining existing basic blocks (typically C++-based) within the same or other top-level (sub)graphs.

### Proposed Actions:

1. Dedicated YAML-Based Nested Graph Definitions
    * Implement dedicated YAML files for defining nested graphs.
    * Store these YAML definitions in the block registry for reuse across multiple graphs.
    * Leverage the `grc-loader` and its YAML-parsing capabilities.

2. Plugin-Like Extensions for User-Defined Nested Graphs
    * Introduce a plugin-style extension allowing users to browse and load nested graph definitions from user-specified directories.
    * This enhances flexibility and modularity in defining and reusing nested graphs.

3. Plugin-Like Extensions for External Graph Sources
    * Extend the plugin architecture to support external sources beyond local files.
    * Include dynamic querying of graph definitions from sources such as databases or community-driven repositories via HTTP(S).
    * Enable nested graph YAML definitions to be fetched and loaded dynamically from external websites or HTTPS-based repositories.

This is a follow-up of the following issues/PRs:
* https://github.com/fair-acc/gnuradio4/pull/479
* https://github.com/fair-acc/opendigitizer/issues/184

RalphSteinhagen, 2025-04-30:
see also: https://github.com/fair-acc/gnuradio4/issues/487

ivan-cukic, 2025-06-25:
3SP so far
- https://github.com/fair-acc/gnuradio4/commit/47f3bef0ab19bea17ce569568f755c1eadc697e1
- https://github.com/fair-acc/gnuradio4/pull/602

iamsergio, 2025-09-17:
1SP: Rebasing pr #650 over ralph's refactorings. conflicts impossible to solve. Rolled back our changes and rewrote them on top of ralph's. Investigated tests crashing/regressions.