fair-acc/gnuradio4 #517: [5SP] CMake integration of block registration macro

RalphSteinhagen, 2025-03-06:
The repository https://github.com/fair-acc/gr4-blocklib-build-concept/ contains a proof-of-concept build system that uses a custom C++ code generator tool to generate code that registers known GR blocks.

The task is to transition this from PoC to production code.

## Context
When blocks are defined, as they are commonly implemented as templates, and specific instantiations need to be dynamically known at runtime, all valid (or all needed) instantiations must be compiled to be loaded at runtime.

Markers such as these are added to the `.hpp` file where the block is defined to define the valid template parametrizations for a block.

This syntax is parsed by this tool https://github.com/fair-acc/gr4-blocklib-build-concept/tree/main/tools which then generates C++ code that registers specific block template instatiations. The syntax is not relevant to this task, but is useful as an example.

```cpp
GR_REGISTER_BLOCK(gr::basic::BlockN, ([T], [U], 3UZ), [ float, double ], [ short, int, long, long long ])
GR_REGISTER_BLOCK("CustomBlockNameN", gr::basic::BlockN, ([T], [U], 4UZ, gr::basic::AlgoImpl2<[T]>), [ short ], [ short ])
```

## CMake integration
The first part CMake needs to do is build the parser tool, which is available at https://github.com/fair-acc/gr4-blocklib-build-concept/tree/main/tools. Review the current implementation and replace it with more idiomatic CMake if required.

After that, the main task is to provide CMake commands that, given a target and a list of header files, trigger the generation of corresponding `.cpp` files, which are then added to the target.

For example, to add blocks defined in `Sine.hpp` and `Adder.hpp` to `libMyBlocks`, one could use this in CMake:

```cmake
gr_generate_block_instantiations(libMyBlocks Sine.hpp Adder.hpp)
```

## Customization

By default, the parsing tool generates a single `.cpp` file that contains all the instantiations defined in the corresponding `.hpp` file. Since that can be complex for the compiler, an alternative mode (`--split` flag for the parser) creates a `.cpp` file for each instantiation separately.

Two approaches should be available to switch this mode on. Passing an option to the gr_generate_block_instantiations command switches on this option for the specified headers:

```cpp
gr_generate_block_instantiations(libMyBlocks Sine.hpp Adder.hpp SPLIT_BLOCK_INSTANTIATIONS On)
```
Defining global `GR_SPLIT_BLOCK_INSTANTIATIONS=On` activates this behaviour by default.

All generated `.cpp` files should be added to the specified target. A build target should be per block-lib module (e.g. `libGnuRadioBasic.so`, `libGnuRadioElectrical.so`, ...) and a globgal one (e.g. `libGnuRadio4.so`). Dynamic llibrary version (other than what's already in the file itself) should be tackled later.

## Integrate with OpenDigitizer

Replace the existing registry block calls in OD with the new system. For non-integral OD blocks, it should generate a static or dynamic library containing the blocks available in OD. Those should then be linked to the UI and service processes. 

The produced `.so` and `.a` libs should be cached on the CI if the GR4 fetchContent hash/tag hasn't changed to minimise recurring block-lib build times.

nicolasfella, 2025-03-10:
Hi,

I started working on this in https://github.com/fair-acc/gr4-blocklib-build-concept/pull/1

RalphSteinhagen, 2025-03-11:
> *sorry for the duplicate - this is for documentation/records keeping. Guess it's better to discuss this here than in the PR*

@nicolasfella thanks a lot for the clean-up and restructuring! Looks very nice.

Would you know whether it's possible to automatically regenerate (i.e. 'delete' and 're-create')  the `.cpp` files in response to changes in the defining `.hpp` file?

The practical use case is that a user 
 * adds or removes a template specialisation (e.g. `[ int ]` -> `[ int double ]` -> `[ float ]`, or
 * changes the template name (e.g. `Block3` -> `Block3a`).
 
 A subsequent build detects the changes but leaves the old (often incorrect) `.cpp` in tact. also just rerunning `cmake` doesn't seem to solve. Could this be solved for regular build calls? :thinking: 
 
 Maybe this question is also for @ivan-cukic

RalphSteinhagen, 2025-03-12:
@nicolasfella job well done! 👍