fair-acc/gnuradio4 #561: win32 test case failure qa_buffer

chrisjohgorman, 2025-04-24:
The following is the ctest output on msys2 / ucrt64 for qa_buffer.  I have a bunch of tests that fail, mostly all are running into a timeout.  This one does not.

```$ ctest.exe -VV -R qa_buffer --timeout=90 --output-on-failure --test-dir build
UpdateCTestConfiguration  from :E:/msys64gr/home/chris/src/gnuradio4/build/DartConfiguration.tcl
Test project E:/msys64gr/home/chris/src/gnuradio4/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: qa_buffer

1: Test command: E:\msys64gr\home\chris\src\gnuradio4\build\core\test\qa_buffer.exe
1: Working Directory: E:/msys64gr/home/chris/src/gnuradio4/build/core/test
1: Environment variable modifications:
1:  GNURADIO4_PLUGIN_DIRECTORIES=set:E:/msys64gr/home/chris/src/gnuradio4/build/core/test/plugins
1: Test timeout computed to be: 90
1: Assertion failed: nSamples <= available() && "Number of required samples is more than number of available samples.", file E:/msys64gr/home/chris/src/gnuradio4/core/include/gnuradio-4.0/CircularBuffer.hpp, line 690
1/1 Test #1: qa_buffer ........................Exit code 0xc0000409
***Exception:   0.21 sec
Assertion failed: nSamples <= available() && "Number of required samples is more than number of available samples.", file E:/msys64gr/home/chris/src/gnuradio4/core/include/gnuradio-4.0/CircularBuffer.hpp, line 690

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.24 sec

The following tests FAILED:
          1 - qa_buffer (Exit code 0xc0000409
)
Errors while running CTest
```

When I run the qa_buffer.exe by hand I get the following output

```
$ E:/msys64gr/home/chris/src/gnuradio4/build/core/test/qa_buffer.exe
BasicConcepts: test(gr::CircularBuffer<int>-AllocatorPortable)
BasicConcepts: test(gr::CircularBuffer<int, 18446744073709551615, gr::ProducerType::Multi>-AllocatorPortable)
iteration 0 - fixed-size data[ 3]: [0, 0, 0]
iteration 0 - full-size  data[ 3]: [0, 0, 0]
iteration 0 - consumed 3 elements - still available: 15
iteration 1 - fixed-size data[ 3]: [0, 0, 0]
iteration 1 - full-size  data[ 3]: [0, 0, 0]
iteration 1 - consumed 3 elements - still available: 12
iteration 2 - fixed-size data[ 3]: [0, 0, 0]
iteration 2 - full-size  data[ 3]: [0, 0, 0]
iteration 2 - consumed 3 elements - still available: 9
iteration 3 - fixed-size data[ 3]: [0, 0, 0]
iteration 3 - full-size  data[ 3]: [0, 0, 0]
iteration 3 - consumed 3 elements - still available: 6
iteration 4 - fixed-size data[ 3]: [0, 0, 0]
iteration 4 - full-size  data[ 3]: [0, 0, 0]
iteration 4 - consumed 3 elements - still available: 3
CircularBufferTest: test(gr::CircularBuffer<int32, 18446744073709551615ull, (gr::ProducerType)0, gr::SleepingWaitStrategy>-AllocatorPortable)
CircularBufferTest: test(gr::CircularBuffer<int32, 18446744073709551615ull, (gr::ProducerType)1, gr::SleepingWaitStrategy>-AllocatorPortable)
Assertion failed: nSamples <= available() && "Number of required samples is more than number of available samples.", file E:/msys64gr/home/chris/src/gnuradio4/core/include/gnuradio-4.0/CircularBuffer.hpp, line 690
```
I will do some investigating on why CircularBufferTest is being passed 18446744073709551615ull under windows and 18446744073709551615ul under Linux.

RalphSteinhagen, 2025-04-24:
@chrisjohgorman thanks for your interest and great that you have some expertise w.r.t. building under Windows. Much appreciated! 👍

chrisjohgorman, 2025-05-06:
qa_buffer.exe exits with a STATUS_STACK_BUFFER_OVERRUN (0xc0000409), when run under msys2 / ucrt64.  

The backtrace identifies 
```
#0  0x00007ffc7647f6fe in ucrtbase!abort () from C:\Windows\System32\ucrtbase.dll
#1  0x00007ffc76481019 in ucrtbase!_get_wpgmptr () from C:\Windows\System32\ucrtbase.dll
#2  0x00007ffc76481201 in ucrtbase!_assert () from C:\Windows\System32\ucrtbase.dll
#3  0x00007ff787c84919 in gr::CircularBuffer<int, 18446744073709551615ull, (gr::ProducerType)1, gr::SleepingWaitStrategy>::Reader<int>::get<(gr::SpanReleasePolicy)2> (this=0x5ff280, nRequested=1)
    at E:/msys64gr/home/chris/src/gnuradio4/core/include/gnuradio-4.0/CircularBuffer.hpp:690
#4  0x00007ff787d35909 in operator()<CircularBufferTestTypes<gr::CircularBuffer<int, 18446744073709551615, (gr::ProducerType)1>, AllocatorPortable> > (__closure=0x5ff3cf)
    at E:/msys64gr/home/chris/src/gnuradio4/core/test/qa_buffer.cpp:387
```

I've tried to examine the binary with gdb, but I can't find the offending code.  I think that when I am examining the code a race condition does not occur, but when I run it on it's own, qa_buffer.exe exits with the error.

To try and figure it out, I ran dr. memory qa_buffer.exe and it confirmed that  I was experiencing a read outside the stack.  Dr. memory's output begins...

```
Dr. Memory version 2.6.20185 build 0 built on Apr 12 2025 03:05:45
Windows version: WinVer=105;Rel=2009;Build=22631;Edition=Core
Dr. Memory results for pid 7792: "qa_buffer.exe"
Application cmdline: "build/core/test/qa_buffer.exe"
Recorded 124 suppression(s) from default E:\msys64gr\home\chris\bin\DrMemory-Windows-2.6.20185\bin64\suppress-default.txt

Error #1: UNADDRESSABLE ACCESS beyond top of stack: reading 0x000000d4af7ff570-0x000000d4af7ff578 8 byte(s)
# 0 ___chkstk_ms 
# 1 _pei386_runtime_relocator
# 2 __tmainCRTStartup
# 3 .l_start     
# 4 ntdll.dll!RtlUserThreadStart           +0x27     (0x00007ffc78d0af58 <ntdll.dll+0x5af58>)
Note: @0:00:00.190 in thread 12448
Note: 0x000000d4af7ff570 refers to 776 byte(s) beyond the top of the stack 0x000000d4af7ff878
Note: instruction: or     $0x0000000000000000 (%rcx) -> (%rcx)

Error #2: UNINITIALIZED READ: reading 0x000000d4af7ff358-0x000000d4af7ff360 8 byte(s)
# 0 std::__cxx11::basic_string<>::_M_check_length
# 1 std::__cxx11::basic_string<>::_M_replace
# 2 std::__cxx11::basic_string<>::assign
# 3 std::__cxx11::basic_string<>::assign<>
# 4 std::__cxx11::basic_string<>::operator=<>
# 5 boost::ext::ut::v2_0_1::reporter_junit<>::on
# 6 boost::ext::ut::v2_0_1::runner<>::run
# 7 boost::ext::ut::v2_0_1::runner<>::~runner
# 8 __tcf_ZN5boost3ext2ut6v2_0_13cfgINS2_8overrideEJEEE
# 9 ucrtbase.dll!_sys_nerr                                           +0x2b2    (0x00007ffc76453a93 <ucrtbase.dll+0x53a93>)
#10 ucrtbase.dll!o_free                                              +0xdd     (0x00007ffc7642042e <ucrtbase.dll+0x2042e>)
#11 ucrtbase.dll!execute_onexit_table                                +0x3c     (0x00007ffc7641dddd <ucrtbase.dll+0x1dddd>)
#12 ucrtbase.dll!exit                                                +0x1a5    (0x00007ffc7641bfc6 <ucrtbase.dll+0x1bfc6>)
#13 ucrtbase.dll!exit                                                +0x149    (0x00007ffc7641bf6a <ucrtbase.dll+0x1bf6a>)
#14 ucrtbase.dll!exit                                                +0x266    (0x00007ffc7641c087 <ucrtbase.dll+0x1c087>)
#15 __tmainCRTStartup
#16 .l_start     
#17 ntdll.dll!RtlUserThreadStart                                     +0x27     (0x00007ffc78d0af58 <ntdll.dll+0x5af58>)
Note: @0:00:00.668 in thread 12448
Note: instruction: cmp    %rax 0xffffffd8(%rbp)
```
I'm going to try to get a build working under clang as I can use the address sanitizer under clang, perhaps that will give me some more insight as to where the code is failing.

iamsergio, 2025-05-07:
No need for address sanitizer, it's failing because of a test assert ` Assertion failed: nSamples <= available()`.
The `STATUS_STACK_BUFFER_OVERRUN (0xc0000409)` is a misleading windows error, there's no buffer overrun

chrisjohgorman, 2025-05-07:
thank you @iamsergio.  I thought the assert was the misleading error.  When I run the program under Linux I get a successful test run, but I get a few assertion failures ...
```
qa_buffer: /home/chris/gr4/src/gnuradio4/core/include/gnuradio-4.0/CircularBuffer.hpp:690: constexpr gr::CircularBuffer<T, SIZE, producerType, TWaitStrategy>::ReaderSpan<U, policy> gr::CircularBuffer<T, SIZE, producerType, TWaitStrategy>::Reader<U>::get(std::size_t) [with gr::SpanReleasePolicy policy = gr::SpanReleasePolicy::ProcessNone; U = int; T = int; long unsigned int SIZE = 18446744073709551615; gr::ProducerType producerType = gr::ProducerType::Single; TWaitStrategy = gr::SleepingWaitStrategy; std::size_t = long unsigned int]: Assertion `nSamples <= available() && "Number of required samples is more than number of available samples."' failed.
```
and the test marches on.  Under windows the assertion occurs and the test ends.  My guess was the producer / consumer loop looped and exhausted the stack, which was why I thought to try the address sanitizer.  I'll try to run a few more gdb runs and examine nSamples and available().  

Thanks again for the pointer.

chrisjohgorman, 2025-05-07:
Okay, this looks like a problem I caused.  In order to get a build I commented out some code in `core/test/qa_buffer.cpp.`  Of course I promptly forgot about it.

Two segments with `aborts` in them 
```
expect(aborts([&reader] { std::ignore = reader.get(1); }));
```
and

```
expect(aborts([&reader] {
    {
                ReaderSpanLike auto inSpan4 = reader.template get<SpanReleasePolicy::Terminate>(3);
                expect(eq(inSpan4.size(), 3UZ));
                expect(!inSpan4.isConsumeRequested());
    }
}));
```
The latter is responsible for the early exit.  Off to find out where aborts is defined.

chrisjohgorman, 2025-05-07:
It looks like aborts is defined in `boost/ut.hpp` but only if the platform supports it, which Windows doesn't.  I have reached the limit of my understanding of things.  There is some guard code on the `expect(aborts([&reader] ... ` code segments in `qa_buffer.cpp` so it is not executed if `__EMSCRIPTEN__` of `NDEBUG` are defined.  Adding a case for `_WIN32` is what I did to get the build to complete.

chrisjohgorman, 2025-06-19:
Okay.  This test case now works on windows. :)  I still need to have a `#if not defined(_WIN32)` around the `expect(aborts(...))` calls, but I now get a working test case.  I strongly suspect the reason for the fix is a working `AtomicBitset.hpp` in windows, but it could very well be attributed to the updated thread Manager too.  Time to make a PR for this.

chrisjohgorman, 2025-07-11:
fixed in 297188f.

closed 2025-07-11 by chrisjohgorman