fair-acc/gnuradio4 #584: win32: qa_ImChart.exe doesn't appear to draw with the right font

chrisjohgorman, 2025-05-23:
Hello All,

First off let me say it's really nice how the tty's have graphs in them like the ones drawn by qa_ImChart.  On Windows the test cases pass, but the font or characters drawn are wrong.

Looking at the first graph of qa_ImChart, all the bits are there.  The color of the graph are correct, the x and y axes are there and the numbers are there.  But...

The lines aren't drawn properly, what should be the x-axis, eg.
```
 ├─────────────┬─────────────┬
```
turns out to be 
```
Γö£ΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓöÇΓ
```
The sine and cosine curves are there, but in the same way the x-axis isn't drawn properly, the braille font used to draw the curves shows up as ...
```
ΓóÇΓúáΓáñΓáñΓúäΓíÇ
```
instead of ...
```
⢀⡴⠚⠙⠲⢤
```
When the curves are drawn using 'X' characters, that is drawn well.

The y-axis suffers from the same problem as the x-axis.  I'm including a text file showing the output of qa_ImChart.exe.  [qa_ImChart.txt](https://github.com/user-attachments/files/20418023/qa_ImChart.txt)

I'm wondering if I could fix this by specifying a specific font in windows?

RalphSteinhagen, 2025-05-24:
Hi @chrisjohgorman — thanks a lot for all your Win32-related reports! These are genuinely helpful, and we appreciate you taking the time to document and share them. 👍

As you may have guessed, Windows hasn’t been our primary focus so far. We're a small team and don’t have any dedicated Windows development expertise in-house. That said, we absolutely don’t want to discourage contributions in this area — quite the opposite. Your input is helping to improve on this, and we’d love to see them tackled with community help.

Would it make sense to collect the various Win32-specific bugs into a single "Windows Compatibility EPIC" issue to keep things manageable? That way, we could list known issues and link individual PRs or sub-issues as fixes come in. It would also help keep Win32-specific discussions separate from cross-platform or toolchain-related topics (e.g. GCC, Clang, Emscripten).

Let us know if you have a better idea for structuring this or if you know other Windows developers 👷 who might be interested in helping out. @mormj @dkozel, do you have any thoughts?

Thanks again for your contributions and persistence — it’s really appreciated!

## 🚧 🚧 🏗  Keep up the good work! 🏗  🚧 🚧

chrisjohgorman, 2025-05-31:
@RalphSteinhagen I'm sorry for taking so long to respond.  I have setup myself with a matrix account so I can converse with you and the leads on the architecture channel.  I will try introducing myself to the architecture group and then see about setting up a meeting time to discuss the best way forward.  Perhaps I can join in on one of the weekly meetings.

I believed I'd found an error that prevented test cases from succeeding and was anxious to chase it down.  It turns out I found and fixed the error, but my test case coverage is still only around 55%. :(  I spent my time pursuing that in hopes I would have some good news to pass on when I got to responding to your post.  

You're welcome for my contributions, and apologies again for my delay in responding.

chrisjohgorman, 2025-06-14:
Okay, I ran into some new information on this bug.  When running in the msys mintty terminal, the bug presents as described.  If I redirect output to a file, the file has the proper formatting.

chrisjohgorman, 2025-06-14:
Hi @RalphSteinhagen to fix this in windows, I need to add an #include <windows.h> to the cpp files and call SetConsoleOutputCP(CP_UTF8);.  I just tested this and it fixes this one. I'll add it to my tree.  Thanks for the pointer to the utf8 problem you had it was a good lead and lead to fixing it. :)