fair-acc/gnuradio4 #632: [5SP] Improve scheduler and messages tests

RalphSteinhagen, 2025-07-17:
* Refactor tests not to manually create std::thread for schedulers,
but to use the default thread pool
  * message_utils contain several really similar functions,
investigate usages and unify as much as possible
(waiting for reply, waiting and read the reply...)
  * tests often rely on the reply message to the sent request
to be the first message that arrives as a response. Or they
ignore all messages that have arrived except the one that
the test needed. As the messaging system is asynchronous,
this is an issue because a message needed by the second part
of the test can arrive and get ignorred while the first part
is being executed.

iamsergio, 2025-07-22:
For the threading:

- [ ] qa_Scheduler
- [x] qa_SchedulerMessages
- [ ] qa_Messages
- [ ] qa_HierBlock.cpp
- [ ] qa_Block

iamsergio, 2025-07-24:
Work done on this ticket:

- round of TSAN debugging and gdb'ing directly in CI to debug races
- https://github.com/fair-acc/gnuradio4/pull/636
- https://github.com/fair-acc/gnuradio4/pull/635 (last commit)
- https://github.com/fair-acc/gnuradio4/pull/638

closed 2025-07-25 by RalphSteinhagen