fair-acc/gnuradio4 #655: cmake fails to run with clang 20

marcusmueller, 2025-09-13:
… bit surprising, tbh, since modern clang is explicitly listed as supported

```shell
mkdir testbuild && cd testbuild
cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -GNinja -DCMAKE_CXX_COMPILER=clang++ -DCMAKE_C_COMPILER=clang ${graph-proto directory}
```

yields

```text
-- The CXX compiler identification is Clang 20.1.8
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib64/ccache/clang++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMake version: 3.31.6
Building Tests and benchmarks.
-- Using /usr/lib64/ccache/clang++ to compile the gnuradio_4_0_parse_registrations tool
-- Configuring parser tool in /home/marcus/src/graph-prototype/testbuild/blocklib_generator/tools_build...
-- Building parser tool in /home/marcus/src/graph-prototype/testbuild/blocklib_generator/tools_build...
CMake Error at blocklib_generator/CMakeLists.txt:52 (message):
  Failed to build gnuradio_4_0_parse_registrations tool:
  /home/marcus/src/graph-prototype/blocklib_generator/tools/parse_registrations.cpp:21:10:
  fatal error: 'algorithm' file not found

     21 | #include <algorithm>
        |          ^~~~~~~~~~~

  1 error generated.

  gmake[3]: ***
  [CMakeFiles/gnuradio_4_0_parse_registrations.dir/build.make:79:
  CMakeFiles/gnuradio_4_0_parse_registrations.dir/parse_registrations.cpp.o]
  Error 1

  gmake[2]: *** [CMakeFiles/Makefile2:87:
  CMakeFiles/gnuradio_4_0_parse_registrations.dir/all] Error 2

  gmake[1]: *** [CMakeFiles/Makefile2:94:
  CMakeFiles/gnuradio_4_0_parse_registrations.dir/rule] Error 2

  gmake: *** [Makefile:124: gnuradio_4_0_parse_registrations] Error 2 [ 50%]
  Building CXX object
  CMakeFiles/gnuradio_4_0_parse_registrations.dir/parse_registrations.cpp.o

-- Configuring incomplete, errors occurred!
```

iamsergio, 2025-09-13:
Do you have `libc++` headers installed ?

iamsergio, 2025-09-14:
the build forces libc++ if clang, the parser tool is just the first to fail if headers are missing