fair-acc/gnuradio4 #656: bundled FFTw

marcusmueller, 2025-09-13:
I understand FFTw is necessary to be bundled for a WASM build (is that correct?), but on any other system, we must use the system FFTw, to be packagable. (Also, nobody wants to be the maintainer for a vendored FFTw)

RalphSteinhagen, 2025-09-15:
No, it is not. Also, it is not the default implementation that is being used.

Unless there are any other objections, I'd remove the FFTw dependency entirely for the `<simd>`-FFT version. 

The latter version is more portable, much faster (factor 2-5),  and has mainstream support from GCC (current SIMD reference implementation) and largely LLVM (via the 'vir-simd' wrapper) as well. Since its pure C++, it has also a better chance of getting integrated w.r.t. GPU acceleration (via SYCL). For reference (CPU only, see also [here](https://github.com/FFTW/fftw3/issues/293#issuecomment-3238974962)):

[image]

marcusmueller, 2025-09-15:
I might be misinterpreting this output, but isn't it slower in all non-power-of-2 settings?

RalphSteinhagen, 2025-09-16:
> I might be misinterpreting this output, but isn't it slower in all non-power-of-2 settings?

The numbers are throughput (normalised to 'N log(N)')-- higher is better:

[image]

RalphSteinhagen, 2025-09-16:
@marcusmueller the non-power of two is realised via the Bluestein's algorithm. Not yet implemented (WIP) for the `<simd>`-FFT version. Stay tuned.